SlimeVR/SlimeVR-Server #1399: Add SlimeVR to Winget

Sacred-Shadow, 2025-04-21:
Microsoft is well on its way to developing a strong and useful package manager with Winget. 
It's becoming increasingly popular and is continuously being improved.

It would be great if the SlimeVR Server could be published there under the name "SlimeVR".
More info: https://github.com/microsoft/winget-pkgs

For many users, this would likely eliminate the need for an auto-updater.

loucass003, 2025-04-21:
i agree winget is nice, we should have releases sent there

ImUrX, 2025-04-21:
You would need to add support for the installer to support silent installs if they are not supported yet, I'm open to anyone maintaining a winget package.

I currently help on flatpak which is outdated and nixpkgs for slimevr