SlimeVR/SlimeVR-Server #1406: Firmware Update with multiple Tracker Crash Server (v0.15.0-rc.1)

Piranja81, 2025-04-25:
With the Firmware update is something wrong when select multiple Tracker to update.
one alone (one by one) works. In my case.

Tldr;
5 Tracker original SlimeVR Tracker; Stock direct out of the box; with v0.4.0.
(10 other original SlimeVR Tracker, already v0.5.4, are connected at the same time to the Server)
I try to update firmware to v0.5.4 wit Slime Server v0.15.0-rc.1.

-> select all 5 Tracker to firmware update. I'm get immediately Slime Server crash (light blue screen of crash / death).

-> select only one Tracker to firmware update. it works.

```
INFO [slimevr] Server found on path: C:\Program Files (x86)\SlimeVR Server
INFO [slimevr] Using Java binary: "C:\Program Files (x86)\SlimeVR Server\jre/bin\java.exe"
INFO [slimevr, webview] Initialized the Sentry client
INFO [slimevr::presence, discord_presence] connected to Discord!
INFO [slimevr] saved window state
INFO [slimevr] send exit to backend
ERROR [slimevr] Java server receiver died
```

```
Board type: SlimeVR,
firmware name: 0.4.0,
protocol version: 17,
mac: F8:B3:B7:80:05:42,
name: udp://F8:B3:B7:80:05:42
2025-04-25 13:29:11 [INFO] [TrackerServer] Handshake received from /192.168.52.212:6969
2025-04-25 13:29:11 [INFO] [TrackerServer] Tracker 11 connected from address /192.168.52.212:6969.
Board type: SlimeVR,
firmware name: 0.4.0,
protocol version: 17,
mac: 24:4C:AB:4A:4D:47,
name: udp://24:4C:AB:4A:4D:47
2025-04-25 13:29:11 [INFO] [TrackerServer] Sensor 0 for udp://F8:B3:B7:80:05:42 status: 1
2025-04-25 13:29:11 [INFO] [TrackerServer] Added sensor 0 for udp://F8:B3:B7:80:05:42, ImuType BNO085, DataType ROTATION, default TrackerPosition null
2025-04-25 13:29:11 [INFO] [TrackerServer] Sensor info for udp://192.168.52.252/0: 1, mag NOT_SUPPORTED
2025-04-25 13:29:12 [INFO] [TrackerServer] Sensor 0 for udp://24:4C:AB:4A:4D:47 status: 1
2025-04-25 13:29:12 [INFO] [TrackerServer] Added sensor 0 for udp://24:4C:AB:4A:4D:47, ImuType BNO085, DataType ROTATION, default TrackerPosition null
2025-04-25 13:29:12 [INFO] [TrackerServer] Sensor info for udp://192.168.52.212/0: 1, mag NOT_SUPPORTED
2025-04-25 13:29:58 [INFO] [HumanSkeleton] Reset: full (WebSocketAPI)
2025-04-25 13:29:58 [INFO] [HumanPoseManager] 0 seconds since last reset. Tracker yaw drifts: udp://08:3A:F2:C3:D5:F4/0 (LEFT_UPPER_ARM), 172.6807 deg (Infinity deg/min) | udp://08:3A:F2:C4:1B:D4/0 (LEFT_LOWER_LEG), 161.7979 deg (Infinity deg/min) | udp://08:3A:F2:C4:1B:D4/1 (LEFT_FOOT), 75.9207 deg (Infinity deg/min) | udp://08:3A:F2:C3:E4:AA/0 (RIGHT_UPPER_LEG), 96.7022 deg (Infinity deg/min) | udp://AC:0B:FB:EE:D9:E3/0 (RIGHT_LOWER_LEG), 4.1471 deg (Infinity deg/min) | udp://AC:0B:FB:EE:D9:E3/1 (RIGHT_FOOT), 95.0134 deg (Infinity deg/min) | udp://AC:0B:FB:F5:88:54/0 (LEFT_UPPER_LEG), 82.2282 deg (Infinity deg/min) | udp://08:3A:F2:C4:1B:26/0 (CHEST), 91.1381 deg (Infinity deg/min) | udp://08:3A:F2:C4:1B:26/1 (HIP), 76.4278 deg (Infinity deg/min) | udp://AC:0B:FB:F1:D3:85/0 (RIGHT_UPPER_ARM), 3.268 deg (Infinity deg/min)
2025-04-25 13:30:46 [INFO] [FirmwareUpdateHandler] Update jobs canceled
2025-04-25 13:30:46 [INFO] [WebSocketAPI] Disconnected: 127.0.0.1, (1000) . Remote: true
2025-04-25 13:31:14 [INFO] VRServer thread interrupted
```

https://github.com/user-attachments/assets/345c3f28-7a50-4f61-9a72-c9b65cc97ab7

whytewuff, 2025-04-26:
There's issues also when the value of the battery resistor is changed from the default.

loucass003, 2025-04-26:
Fixed with RC.2

whytewuff, 2025-04-26:
Nope, I tried with RC.2 just now, it's definitely not fixed. I'm using another flashing tool meanwhile.

loucass003, 2025-04-26:
Can you please re record with rc2 and open the console CTRL + SHIFT + I to see the logs

whytewuff, 2025-04-26:
Bad request to the external API. Here's the body:
```
{"boardConfig":{"type":"BOARD_WEMOSD1MINI","ledPin":"2","enableLed":true,"ledInverted":true,"batteryPin":"A0","batteryType":"BAT_EXTERNAL","batteryResistances":["200",100,220]},"imusConfig":[{"intPin":"D5","sclPin":"D1","sdaPin":"D2","optional":false,"type":"IMU_BNO085","rotation":180},{"intPin":"D6","sclPin":"D1","sdaPin":"D2","optional":true,"type":"IMU_BNO085","rotation":180}],"version":"SlimeVR/v0.5.4"}
```

And the response
```
{
    "path": "$input.boardConfig.batteryResistances[0]",
    "reason": "Error on typia.assert(): invalid type on $input.boardConfig.batteryResistances[0], expect to be number",
    "expected": "number",
    "value": "200",
    "message": "Request body data is not following the promised type."
}
```

Looks like a type issue.

Idk why it won't recognize the second tracker, it's there indeed.
Thanks!

loucass003, 2025-04-26:
This is a different error tho. This is not even the same page as the first recording

PR #1413 (2025-04-28): Fix battery resistances being sent as string instead of numbers
Fixes: #1406
closed 2025-04-28 by loucass003