SlimeVR/SlimeVR-Server #1423: Add a Set WiFi button to the Serial Console to allow the user to send an SSID and Password to a specific port device

Oldrego, 2025-05-05:
Currently, if another port device is installed on the user's machine, SlimeVR-Server will attempt to send Wi-Fi Credentials to that device. If it receives no response from the device, then it will wait for a response from that device indefinitely, or otherwise repeatedly attempt to send the Wi-Fi Credentials erroneously. The only current solution to this is to uninstall the port device, as disabling the device's drivers doesn't prevent this behavior from happening.

It would be useful for a button to exist on the Serial Console window for setting the Wi-Fi Credentials. Upon clicking it, the user is prompted to enter the Wi-Fi name (SSID) and password. Then the serial console will send a `-> SET WIFI "<SSID>" "<PASS>"` command to the device currently selected in the serial console. I've mocked up what this might look like below.

[image]

This would easily allow users to bypass any issues related to this.

Eirenliel, 2025-05-05:
The correct solution would be to send wifi credentials to all ports compatible ports, it will also allow multiple slimes to be provisioned at the same time.

Oldrego, 2025-05-06:
It would also be useful for the user to have the option to send `-> SET WIFI "<SSID>" "<PASSW>"` to a specific port device, although I also would recommend your solution to also be implemented as well.

gorbit99, 2025-05-06:
Honestly allowing the user to run commands in that window wouldn't be a horrible idea. They could even be warned if they are trying to do something possibly unwanted, like factory reset the tracker

ButterscotchV, 2025-06-30:
See https://github.com/SlimeVR/SlimeVR-Server/issues/1122

closed 2025-11-19 by loucass003