SlimeVR/SlimeVR-Server #1427: Display Issue Due to Translation

Sacred-Shadow, 2025-05-07:
The main menu shows a scrollbar for me when set to German,
apparently because the word “Körperproportionen” isn’t automatically wrapped.
Either the language file needs to be fixed, or the menu should have a fixed scale to wrap text automatically.

[image]

loucass003, 2025-05-14:
Will be fixed with a new feature soon. The icons wont have text under and have tooltips instead. Kinda like discord server list.

ImUrX, 2025-05-16:
I dont really see why we should make the text appear on hover, it's already fixed with #1362 

[image]

Sacred-Shadow, 2025-05-19:
Fixed in 0.15.0

closed 2025-05-19 by Sacred-Shadow