SlimeVR/SlimeVR-Server #1433: New wifi provisioning

loucass003, 2025-05-12:
Redesign the wifi provisioning page completely

Need to work for all of those three cases:
  - Firmware version is older than the first version that added wifi scan
  - Firmware verison older than the one that added tracker to tracker provisioning
  - Firmware version that has tracker to tracker provisining

The flow of the provisioning would look like

- Ask the user to connect a tracker
- Show a dropdown with all the found ssids from the serial command of the tracker (Change to a text field if the connected tracker is older than wifi scan)
  - Have the input password underneeth
- Show the provisioniong progress.
- If connected tracker versions is older than Tracker to Tracker provisioning:
   - Prompt the user to conect their next tracker.
   - Make sure to remember SSID and password from first step / reuse them for the next tracker
- if tracker to tracker provisionign is working
  - Have a visual feedback that other trackers are provisioning
  - Still prompt that the user can plug other trackers if they like to
- In the future we will know what type of set they have and be able to tell that the provisioning is complete and that they can go to the next step

gorbit99, 2025-05-12:
Tracker-Tracker provisioning PR: https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/436

Polymoria, 2025-05-13:
Awesome ticket, a lot of much-needed functions, don't have anything to add so far, so added emoji and fixed some typo 💃

evogel42, 2025-05-19:
👋 Here's the link to the figma of the design for this page, approved by Futura :
[Tracker connection page figma](https://www.figma.com/design/np5svLs6dH1UbPGPwVlwAW/SlimeVR-Server---Tracker-connection-page?node-id=0-1&p=f&t=L9P8LJbF7m8Rja28-0)

If there are any questions or problems please let me know ^^

Bluebie, 2025-06-21:
Maybe this page could show a list of unprovisioned trackers discovered through Tracker-Tracker provisioning, and offer users a button for each to manually connect them, so transmitting wifi credentials over not-very-encrypted radio is a user action rather than a passive automatic default? It could feel nice to click each and have a little progress thing as each tracker connects. It could also be nice in conference or meetup environments to ensure only the trackers you intend to connect are provisioned and not any in nearby areas.

Polymoria, 2025-08-08:
I unasigned it from Gorb and asigned to Maya :3