SlimeVR/SlimeVR-Server #1434: Add an ability to control smol slime and smol slime dongle via server serial

Eirenliel, 2025-05-14:
Right now both smol slimes and smol slime dongles are completely ignored by server's serial console. They have unique PIDs, we can easily detect them and use them too!

For both of them we need to be able to read serial logs. We also need to be able to send a few important commands, like reboot and pair. Only correct commands need to be shown for the correct device, i.e. no wifi scan etc needs to be show for smol slimes, and no pair command for big slimes.

sctanf, 2025-05-14:
as the receivers are unique, each serial device needs to be associated to them, and controls should be eventually be shown, if available, per tracker as well
the receiver should accept commands for itself or commands for a tracker, by its serial number, and forward/receive from the tracker
there may need to be a polling interface so the server knows what state the receiver is in, and can reflect the state (ie pairing, how many trackers/ which serial numbers or known trackers, if the server is able to associate them, are remembered)

Eirenliel, 2025-05-14:
This is not what this ticket is about. I'm talking purely to be able to send console commands via thing that already exists in the server - serial console screen. It doesn't know about anything, it just shows serial and can send serial. All of the stuff you mentioned is cool and is needed, but it's not what this ticket is about, sorry.

sctanf, 2025-05-14:
ah i see, i assumed this was a good place to track these; only thing i should mention then is the "exit" command is added to leave pairing without a reboot