SlimeVR/SlimeVR-Server #1505: Freeze rotation when tracker disconnects

art0007i, 2025-07-26:
Currently if any of my trackers disconnect they just end up going to a default standing pose, which looks very jarring if for example my chest tracker disconnects for a few seconds while laying down (pretty common occurrence for me). I think it would would be way better if the server remembered the last rotation that the tracker was in and used that until it reconnects.

SummerSigh, 2025-07-26:
My idea is simple, but should probably go in the firmware repo. Fusion should continue to run on the device if it has lost connection with the server for some duration (maybe 5 minutes?) so that if it happens to reconnect  with the server within that time, the fusion doesn't use old data and therefore you don't need to do a full reset in the server.

ButterscotchV, 2025-07-26:
This used to be the original behaviour, but was changed in favour of falling back to the nearest tracker (as if you didn't have any connected). We could bring back similar behaviour in a better way through allowing you to make the timeout time longer? You could in theory make the timeout time as long as you want and it would retain the rotation for that time.

Alternatively, we could also make it so you can configure whether disconnected trackers are retained in their skeleton slots. I'd like the timeout option a lot better.

We should also probably work on the interpolation at some point so we can kinda estimate the poses better even just with leg trackers so it won't just stick straight down.

art0007i, 2025-07-26:
Is the current behavior useful in any way? Personally I don't see a reason why I'd want my trackers to point straight down whenever they disconnect ever, if they dropped wifi connection for a few seconds or if they just ran out of battery I think I'd always want to retain the rotation of the tracker how it was before the disconnect.

Is it just to make it more obvious that the tracker disconnected?

Either way an option to increase the timeout would work perfectly for me, since like you say I could just increase it to an arbitrarily large value.

ButterscotchV, 2025-07-26:
> Is the current behavior useful in any way? Personally I don't see a reason why I'd want my trackers to point straight down whenever they disconnect ever, if they dropped wifi connection for a few [...]

The current behaviour is useful because it just treats the tracker as not existing and continues with the data it does have available. It's an unknown for when it will come back, so the timeout acts as the retaining factor, and it is assumed it will not return if it's still disconnected after that period of time.

Ofc sticking straight down is not ideal, usually it would be ending up dropping like hip and reverting to chest only, this is just a bit of an unfortunate scenario that there's no other tracker to fall back to. The timeout time could def be configured pretty easily so imo that is the best option.

art0007i, 2025-07-26:
ahh ok, thanks for explaining!