SlimeVR/SlimeVR-Server #1510: [Feature Request] Machine learning for tracking algorithms

Znunu, 2025-07-30:
Because of the inherent nature of IMU tracking, slimes have lots of issues with tracking. From my understanding, various algorithms exist in the code that attempt to fix this (e.g. the recent stay aligned update, the various calibration poses etc). Yet none fix the issue entirely and they all rely on manual tuning of various parameters. 

To me, this sounds like a problem where ML would be worth trying. Especially with so much training data available, from many people using slimeVR. Even without a true reference available for this data, it might still be possible to figure something out.

Eirenliel, 2025-07-30:
I'm sorry, but this is not a "feature request", it's a "i have an idea, why we don't throw a neural network at it?", which is something that everyone have thought of already, and something that might be attempted in the future when we have resources or someone will have cool ideas and desire to implement it.

Unless you have more concrete ideas to try besides "ML", like algorithms, white papers, or full solutions, or you want to research this topic yourself, I'm gonna close this issue.

Znunu, 2025-07-30:
Well, I hope to see better algorithms someday, whatever techniques it requires. Perhaps it would be better to continue this conversation on Discord?

closed 2025-07-30 by Eirenliel