SlimeVR/SlimeVR-Server #1538: [Feature request] Track battery status of past connected trackers.

fuka-oni, 2025-09-06:
It would be nice if trackers already connected kept a battery history. 
Similar to how trackers appear grayed out when disconnected, but in the next session with last battery status. 
(also could add other info like last session or something that is interesting)

This way, I wouldn't have to connect all the trackers every time I want to charge the ones that are low on battery. 
It would be nice if it was a toggleable option, like: "keep history and information from previously connected trackers"

viniciusvieirafx, 2025-10-15:
that would be very useful and easy to implement, I think it would be a good addition

loucass003, 2025-10-30:
Funny enough we actually already keep the last battery value. We just hide it in the ui if the tracker has a disconnected status.

Yk what ... that actually does not look too bad

[image]